tebl/RC6502-Apple-1-Replica #2: Forum?

davischris, 2019-03-19:
Is there a forum for discussion on this project?  I have some questions.

tebl, 2019-03-30:
Don't have one, thus far I'm assuming your tinkering with it makes us two :-)

Not sure if the reply to your blogger comment made it through to your end, but sent some initial troubleshooting tips to start with on it. Pasting the same thing here, just in case the reply was lost to the ether:

> Haven't gotten around to getting a revision F produced yet, but did notice a trace missing on some previous revisions. Added a note with picture on https://github.com/tebl/RC6502-Apple-1-Replica/blob/master/RC6502%20Apple%201%20SBC/README.md, check that you have continuity between these points.
> 
> Check that you're getting a generated phi0 and phi2 on the cpu, preferably on a scope though a voltmeter should show it being alive by measuring in the middle (ie not 0v or 5v). RDY and SYNC should show similarly if I remember correctly, this would show that the cpu is running code.
> 
> Double-check that it's jumpered correctly with PIA, ROM, RAM to enabled (1-2 I believe), there's also a link below the oscillator that was added to the later revisions. ROM jumpered according to size, note 28C64/28C256 - 27C64/27C256 have a different pinout.
> 
> Let me know what you find out :-)

davischris, 2019-04-01:
I have it working.  I suspect it was two issues: First, I was mixing 74LS chips with 74HCT.  I just made sure I used three 74HCT chips.  Also, it does not appear to work with a Rockwell R6502AP (or I have a bad one.)  It does work with an SY6502 (pulled from an Apple II) and I also have it working with a Rockwell R6502A.  Thank you very much for your work on this.  Be encouraged - I had to make 5 boards (the minimum order) so there are four of my friends also working on it!

jjohnson873, 2019-04-18:
Hi Chris. Just finished soldering all the passive components and the sockets on to the board you sent to me.  I've got all of the IC's, but I'm waiting on the SY6502, and the 6521 from China.   The AT28C64 should be in shortly.   I'm also waiting on an EEPROM programmer from China as well.

closed 2019-06-23 by tebl

0cjs, 2020-01-11:
> Also, it does not appear to work with a Rockwell R6502AP (or I have a bad one.) It does work with ...Rockwell R6502A.

Does anybody have any further information on why it might not work (or whether it does actually work) with the Rockwell R6502AP, and could this possibly also apply to the R6502P? I have just built a kit (bought from an auction site) using a revision F+ board. I can't tell you if it's exactly the board from this repo, but it came with parts from a guy who appears to have built his own and got it working, so I'd be surprised if he's selling a known-bad design or parts known not to be working on the board.

And what is the difference between the P and non-P versions anyway? I had assumed that the "P" just meant a plastic (as opposed to ceramic) package, but my part looks like it has a date code of "1349," by which point I thought nobody was using ceramic packaging for 6502s any more. (And not manufacturing NMOS ones either, for that matter, but who knows? The complete markings on the chip are, `R6502P  13500-13  MEXICO  1349  B50323-8`, and a logo.