tebl/RC6502-Apple-1-Replica #3: No life...

rockcat, 2019-09-23:
Hi
Are there any known issues with your rev F board?
I've built one but there's no sign of life. All soldering looks ok but I may just have a short somewhere...

Thanks
[image]
[image]

rockcat, 2019-10-04:
HI
Found the problem - a dodgy capacitor - replaced it and now no short.
Next problem is that I  am getting millions of newlines thrown at me.

Also, do you have the source code of the ROM?

Thanks

rockcat, 2019-10-14:
All working now :)
My 7400 had got fried.
Thanks for the PCB design !

tebl, 2019-10-20:
Been sick a while, so I didn't notice the ticket - glad to see you got it working. Never imagined the 6522 working directly without changes though, so that's new :-)

I used the ROM contents out of the Ben Heck project (https://github.com/thebenheckshow/158-tbhs-apple-1-replica/tree/master/software/code), it also has the source file. That project never had any hardware documentation, which is where mine got started to fill the gaps and have fun along the way.

closed 2019-10-22 by tebl