tebl/RC6502-Apple-1-Replica #6: Can't compile the arduino sketch. Matching lib needed.

Creep69, 2020-02-16:
Hello,
I am also trying to get the board running. My problem is the Arduino sketch for the serial communication. To compile it, you need to add a MCP23S7 lib. But the matching lib is not provided in the github project. I tried at least 5 different libs, I found on the internet. Also one lib another guy sent me who already built the Apple replica. But no lib matches. The class itself needs different parameters, the methods have different names (and parameters). Is there a chance you could give a link to a suitable library? My board is complete, but the Arduino is a showstopper.
Best regards,
Rene

tebl, 2020-02-17:
Sounds like I'm overdue to revisit the Arduino sketch and update things, but in the meantime I've moved things around and in the lib-folder at https://github.com/tebl/RC6502-Apple-1-Replica/tree/master/software/arduino/PIA%20Communicator you'll find a copy of the library I've been using. Sketch itself in the pia_communicator folder as previously.

Creep69, 2020-02-17:
Great, thanks a lot for the immediate fix. Now, the sketch is working fine.
Rene

closed 2020-02-17 by Creep69