tebl/RC6502-Apple-1-Replica #9: RC6502 Rev D RAM and ROM enabled jumpers

cskordis, 2020-04-25:
I have built the RC6502 SBC Rev D plugged into the backplane.
So far the arduino terminal coms works but no backslash.
I was wondering what the RAM_EN and ROM_EN should be set to on the SBC.

Regards

Costas

  
[image]
[image]

0cjs, 2020-04-28:
`RAM_EN` should be on the left-hand pair of pins to enable the RAM, and `ROM_EN` should be on the top pair of pins to enable the ROM. You have both these disabled in your photograph. You also of course need to make sure that the  ROM is correctly programmed with at least the Apple 1 Monitor (WozMon) programmed into the highest 256 bytes of your 8 KB EEPROM.

Since you appear to be using an AT28C64 8 KB EEPROM, the A13 and A14 jumpers should be left disconnected. These are used to select which 8 KB bank would be used in a 32 KB AT28C256 EEPROM. If you ever switch to one of those, you'd want to program the lowest 8 KB bank and jumper the right-hand pair of pins for both.

You can find more information on jumper settings (and perhaps some other useful information) in [my notes here](https://github.com/0cjs/sedoc/blob/master/8bit/apple1/replica-sbc.md).

cskordis, 2020-04-28:
Thank you Curt

I used the 8k image from the github site on the 28C64.
But I am not getting the back slash character

Regards,

Costas Skordis

0cjs, 2020-04-29:
I would expect you're not getting the backslash character because, according to your photographs, you have incorrect jumper settings.

Also, if you can please trim down your responses it will make them a lot easier to read. ([View your comment on GitHub](https://github.com/tebl/RC6502-Apple-1-Replica/issues/9#issuecomment-620660330) and you'll see your two-line response is buried in fifty or so lines of mostly irrelevant stuff, including a complete copy of my comment immediately above.)

cskordis, 2020-04-29:
Thanks Curt.
I did set the jumpers as specified but I don't get a response only from the arduino.
Interesting it looks like A15 is not pulsing but constantly high.
I don't know of the 8k ROM image is the right one although I did download it from the github
I've ordered new MCP23S17 and 6821 just in case the ones I have are fake. I know the CPU, RAM and ROM ICs are good. 
I've also order Rev H of the SBC board and will build that one in case the Rev D board has issues.

Costas

0cjs, 2020-04-29:
Ok, if your jumpers are now correct there is likely to be a fair amount of back-and-forth in the debugging process. I suggest creating a new post in [the 6502.org Forum](http://forum.6502.org/) about this, since there's a lot more people reading that and you'll get more detailed help. (I'm not sure if anybody else besides me there is using this particular board, but at this point the problems can be addressed by a  fairly generic 6502 SBC debugging process.)

You'll want to reference this issue and also explain in detail what test equipment you have available (e.g., do you have an oscilloscope?) and the particular debugging steps you've taken so far. The two obvious places to start are to see if the CPU appears to be starting up correctly and looping correctly in `GETLINE` while waiting for input and if the input and output chars are appearing correctly on ports B and A of the PIA.

> I don't know of the 8k ROM image is the right one although I did download it from the github

It's probably a good idea to read back your ROM and confirm that the code in the top 256 bytes of the ROM matches the [code from the Apple 1 manual](https://archive.org/details/Apple1Manual1976/page/n6/mode/1up). You can even do this by wiring up the ROM on a breadboard with the appropriate pull-ups and -downs and seeing what its putting on its data pins. Checking the reset vector at $FFFC, the first few bytes of the initialization routine at $FF00, and `NEXTCHAR` at $FF29 should probably be fine.

> Interesting it looks like A15 is not pulsing but constantly high.

That would be expected if it's looping at `NEXTCHAR` ($FF29-$FF2D), waiting for input, since all the addresses it's reading are in the upper 32K of RAM.

> I've also order Rev H of the SBC board and will build that one in case the Rev D board has issues.

Right. I'm not familiar with the D board; mine is an "F+".

cskordis, 2020-05-03:
Curt,

I think I found the issue.  My eBay purchase of the MCP23S17 gave me a MCP23017. Never trust eBay  sellers.

Have ordered them from DigiKey.

Will let you know how I get on.

Costas

cskordis, 2020-05-23:
CJ,

I hope you are keeping safe.

Just wanted to let you know that I have 2 RC6502 running perfectly. Thank you for all your help.

I’ve just ordered the RC6502 Terminal and was wondering where I can find the firmware was for the ATMega 328P Video and Keyboard.

Regards

Costas Skordis

tebl, 2020-05-23:
Funny story about the firmware for that, there was a pandemic... PCBs got
delayed and I'm in the process of moving, so I haven't gotten started on it
yet.