tebl/RC6502-Apple-1-Replica #15: SBC Schematics seems incorrect

hugoferreira, 2020-12-18:
Pin 6 of the Arduino Nano is reported to be connected to the CLOCK (which would be interesting, for those that don't have an available crystal, or to do some cycle-stepping). However, at least in the Revision H of the PCB, this pin is not connected.

0cjs, 2020-12-19:
It's not possible to slow/pause the clock to cycle-step an NMOS 6502, or run it at much less than about 100 kHz. ([Reference](http://wilsonminesco.com/6502primer/ClkGen.html)) Unlike CMOS 65C02s, the NMOS versions do not have a static core and will lose the data in their registers if the clock is too slow. (You can stop the clock with CMOS 6502s, so long as you do so only when φ2 is high.) The standard way of cycle- or instruction-stepping a 6502 is to halt the CPU by deasserting RDY.

There might also be issues with the 6820/6821/6521, since it has a clock input.

That said, yes, the pin might be useful for clock generation, though it would increase the complexity of the software on the Nano.

hugoferreira, 2020-12-19:
I agree with everything you've said. That's why I'm reporting this as a bug :) The clock label in the schematics PDF should not exist. 

[image]

0cjs, 2020-12-19:
Ah, I didn't realize you were saying the label should be dropped! Yes, I agree.