tebl/RC6502-Apple-1-Replica #16: Video Out gets stuck and other Instabilities

hugoferreira, 2020-12-19:
First of all, I'm not sure if this is an issue, or some problem of my build. Nonetheless, I hope someone can help me out (and in the process, maybe document some learning).

**TL;DR;** PIA Communicator either gets stuck, or enters and endless loop of displaying a mix of garbage or the same character. One way to force this in my build is to simply connect the BUS to a breadboard.

As far as I understand the Apple 1 Video Out, there are two signals, RDA and DA. DA says that there's a character to be displayer, and RDA informs the PIA that the terminal is ready to accept that character. Once RDA goes from HIGH to LOW, then DA becomes low. DA is connected to PB7 in the PIA, and is the only pin on PB that is configured as input. The transition from HIGH to LOW in DA is controlled by the negated output of CB2.

Now, whenever the SBC becomes "stuck", according to my quick debug, I can see two things: first, the CPU is not stalled. The Address lines are changing, as well as the data lines. But there's an inconsistency in the signals. CB2 is outputting LOW, which is negated by the 74xx04 IC. I can confirm that its output is HIGH (4.62V). I can also confirm that one pin of the 10nF capacitor exhibits 4.62, but the other pin is at 0V. This is confirmed by probing PB7 which is LOW (should be HIGH). Something, somewhere, is asserting this line LOW, when it shouldn't.

Now, from the schematics, this line seems to be connected solely between the Arduino and the PIA. Nothing else is in between. So my hypothesis is that either the PIA changed the DDR of PB somehow, or the Arduino is pulling the line LOW (somehow).

By the way, I can stall my system in the following simple way: simply connecting the BUS to a breadboard 😕

jpdias, 2020-12-19:
I also did build the same version of the RC6502 and I'm having an even worse experience since I'm not obtaining any meaningful output from the Arduino. I already removed/inserted most of the ICs and even changed some with some available spares. No change in behavior. The attached image shows the output, and it keeps going forever like so.

[image]

hugoferreira, 2020-12-19:
It seems that stabilising potential current spikes with a 330uF electrolytic capacitor (best place I've found is near the BUS connector on the bottom), solved the problem (at least it has been running as expected for the last 50 minutes straight).

Now, concerning that video out fails when I connect the SBC to a breadboard, I'm starting to suspect might be due to me using version S of the PIA (W65C21S), which does not have limiting resistors. My main hypothesis is that connecting to the breadboard elongates the signal lines into transmission lines, which might lead to signal reflections if not correctly terminated. The limiting resistors of the N version might act as line terminators on the near end. But this is all wild guess, until I receive a batch of W65C21N's.

tebl, 2020-12-22:
I've done very little work with W65C series chips, so as it stands - I consider those a theoretical might work at the moment. From what I remember of the documentation, the N chip is the one that can be used without changes to the circuit.

There's been quite a lot of fake chips of both variants floating around, so if they came from ebay/AliExpress and similar sites it may be a good idea to try out a second chip - I've mostly had luck with MC6821/R6502.

hugoferreira, 2020-12-22:
These came directly from Farnell, so I assume it's a reliable source 👍 As expected using an old Rockwell 6502 solves the PHI2O problem, which is consistent to my hypothesis on the sensitivity in the line. Now here comes the odd part: with a W65C02, _just probing_ the PHI2O pin with the scope is enough to block the video out, and only a reset solves the problem. Still not sure if this is because the PIA is being picky and if a 65C21N would solve it... But it is funny, isn't it? My nexts steps are going to be:

a. Try a 50 Ohm near-end termination on the PHI2O;
b. Prolly just cut the trace of the PHI2O and put a bodge wire between the PHI2 and the E/PHI2 on the PIA.

By the way, I know this is the way it's presented in Owad's book, but was there any particular reason you had in choosing to expose PHI2O instead of the oscillator's CLK directly on the BUS?