tebl/RC6502-Apple-1-Replica #21: How to write to digital I/O LEDs?

liamoc, 2021-07-09:
Hi, thanks for a great SBC. I've assembled a number of additional components like the Joystick/Digital I/O board and the VDU too. So far, everything works except I don't know how to write to the LEDs.

I tried using WozMon to write to the same address I can read the buttons from (`A000`) but nothing happens -- the LEDs are just always on. Is there some direction register I have to set or something?

tebl, 2021-07-09:
It should be fine to just do a write to that address, but I seem to
remember that there was something about WozMon itself that was the issue
(or a combination with something I did). Maybe try a small machine code
program or something in basic.

liamoc, 2021-07-09:
Hmm, I tried running this in BASIC:
```
> POKE -24576, 0  
> POKE -24576, 255 
```

And the lights remained solidly on. I think `-24576` is the correct location for `$A000` right? I'll try a machine code program soon.

tebl, 2021-07-10:
Now that I'm looking at the schematic, I think that the 373 should be a
374. They have the same pinouts and may have exchanged it on my board
without updating the document.

liamoc, 2021-07-10:
Alright, I've ordered some 374's. When they arrive I'll see what happens.

liamoc, 2021-07-21:
This worked! So it should be 74LS374 rather than 74LS373.

closed 2021-07-21 by liamoc