tebl/RC6502-Apple-1-Replica #22: Reducing noise in video output?

liamoc, 2021-07-13:
Hi, I recently built a VDU and both amplifiers. With either amplifiers I am getting a flickering pattern of horizontal lines, clearly visible particularly in the background. I've got some old micros that produce similar noise, so I assumed it was normal, but the screenshots in the gallery in the repo don't seem to show the same pattern.

I've attached photos in various graphics modes to demonstrate what i mean. Does anyone know which component might be at fault here? It'd be nice to get a clearer picture.

[image]

[image]

tebl, 2021-07-14:
I have Dragon 32 with a dreadful picture as well, which is something I
expect is something of quirk for the chip. The circuit was designed when
CRTs was the norm, so it might not have been an issue back then.

For the most part, working with the analog stuff is a bit out of my reach
though there is a few tips I can provide. The first is to use a good
shielded cable, then check the TV settings; turning the colour down until
you get B/W and if the lines disappear they are in the colour circuit (turn
down any sharpness setting to avoid TV messing with the signal).

liamoc, 2021-07-21:
Alright, thanks for the tips!

closed 2021-07-21 by liamoc