tebl/RC6502-Apple-1-Replica #26: Why is the peripheral clock (φ2out, bus pin 23) separate from the CPU clock (φ2in, bus pin 21)?

0cjs, 2021-08-14:
The 6502 has a φ2out pin (39) to allow access to the system clock when the internal oscillator is used. This doesn't seem necessary if an external oscillator is used since it can be fed to the 6502 φ2in pin (37) and also any peripherals that need a clock signal. At a quick glance this seems to be standard practice: the Apple 1, Apple II, Apple IIc and Acorn Electron for example leave CPU φ2out floating and send the same φ2 signal to the CPU and all other peripherals, if I'm reading the schematics right. (The Apple 1 is slightly confusing due to a third clock signal "φ0" on the expansion connector; it's not clear to me what that does, but it does not supply clock to the 6820.) Further, MOS themselves often did not provide a φ2out pin on variants supporting only external clock, including the (discontinued for legal reasons) 6501 and the Commodore 64's 6510.

Is there a particular reason that this design (on both the CPU board and the SBC) uses φ2out from the CPU for the peripherals rather than the same clock signal that's sent to the CPU? Or is it just what seemed right or obvious at the time? I have the feeling that this is incorrect (and also thus wasting one of the bus lines), but I do not know enough about this to be sure.

tebl, 2021-08-14:
I think it is mostly due to the RC6502 was my first attempt at building a
computer, but mostly based on the Apple1 Replica creation book while
attempting to make it fit as well as I could with the RC2014.

Apple1 had phi0 as the external clock source the CPU, buffered it is Phi1as
used internally by the CPU while delayed into Phi2 for external components.
The confusion does start to crop up when the later 65c02 came about as it
starts using the Phi1 clock in general instead, I assume because the static
design did not need the phase delay.

There's a significant chance I'm still mixing things up here, but I think
mostly I just put them there just in case I needed them later.

0cjs, 2021-08-15:
From looking at the schematics, it doesn't look to me as if the Φ₂ output on the CPU was ever used. Here's how it works, as best as I can determine so far. (This is based only on the schematic from the back of the _Operations Manual_; I don't have any real boards or anything like that to look at.

1. Clock is generated on the video side and brought to the computer side at two places on the schematic, labeled `CL`.
2. One `CL` comes in the 74157 B7 at the upper left, goes down and right through the `"6502"` solder bridge pad in the 6800 clock area (surrounded by a dashed line and unused with a 6502) and up to the CPU, connecting to pins 36 (unused on 6502) and 37 (Φ₀ on MOS 6502,  Φ₂ in on W65C02, in either case the input for an external clock. The labelling beside the pin is a bit confusing, however, as it's `Φ₀,Φ₂` and pin 39 is indicated nowhere on the diagram, but connecting pin 37 and 39 is obviously incorrect, so I assume that pin 39 was just left floating, as you might well do with an unused output, and thus nothing is clocked from that.
3. That same `CL` line also goes down past the `"6502"` solder bridge pad to the `E` input (pin 25) of the 6820, indicating that's clocked by `CL`, and that also goes to the far right side of the diagram to the `Φ₂` pin on the expansion bus edge connector. There's also a `Φ₀` pin on that connector which is fed from the second `CL` box, so both `Φ₀` and `Φ₂` there are nearly same clock, only `Φ₂` is gated by the 74157 at the upper left and `Φ₀` isn't. (I don't know the reason for this off-hand.)

I trace through much of that above just to try to make sure that the pin 39 output isn't actually being used, and all peripherals use the same clock source that the CPU is using, rather than the CPU's output.

So I'm thinking that the replica should be doing the same thing, right?

0cjs, 2021-08-16:
I just dropped some probes on the bus pins of my SBC to see how pin 21 `Φ₂` (yellow), pin 19 `Φ₀` (cyan) and pin 23 `Φ₁` (pink) compare. It turns out there's nothing on bus pin 23, so I moved the probe to 6502 pin 3, where `Φ₁` out is obviously working. I'm guessing I might have missed something in my build, but I'm not really feeling like debugging it right this minute since it's not that important.

What is interesting is to look at the difference between the clock from the external generator on pin 21 and the output of the CPU (an old Rockwell NMOS part) on pin 19:

[image]

That is not nearly as good looking a clock; as well as the slow ramp-up it also for some reason doesn't quite come down to ground. (I carefully confirmed that both channels are using the same offset voltage/vertical setting, and both probes are taking their ground from a short wire crimped to a dupont connector on pin 17.) I suspect that this is more evidence that we should not be using the 6502's clock output (unless we were using its clock generator), but again, I'm not an expert on this stuff.

tebl, 2021-08-16:
I don't have any actual boards to look at either so I couldn't say either
way, but guides for 6502 seem to have it the same way so this may be a
difference from being a work-alike vs a direct replica.

Wozniak has shown a tendency to break the rules somewhat, mostly by having
complete control over the HW and working the minute specifics of it.

The important part of Phi2 is ensuring RAM (or other non-65xx chips) writes
are done when the address lines have stabilized, the signal in question is
itself is a slightly delayed version of Phi0 - so a delay elsewhere might
result in something that would still work the same way. Given that the
source of the clock is elsewhere, this also seems to solve the secondary
problem of buffering the clock. So he probably did not need it.

In practice, it might work as long as the timings match up correctly. The
scope view is interesting, but I suspect it is degrading somewhat due to a
small capacitor added to delay it - probably in part a reason why using the
same internal and external clock on pin 37 became the standard once the
65c02 and in particular the higher speeds of the WDC 65c02s came about
(going to static registers instead of dynamic ones might also be a factor
for why it was simplified).

0cjs, 2021-08-17:
Hmm. Which guides are these? I got [this from 6502.org](http://forum.6502.org/viewtopic.php?t=6689#p85264):

> ...the system clock (phi2) with the W65C02 should be the same signal as is going to into Pin 37 of the CPU, not the Phi2O coming out of pin 39 as it was in the old NMOS days. In fact, in my experience the old NMOS systems work better this way too. At least they seem to overclock better.

Then again I see that [Grant Searle's design](http://searle.x10host.com/6502/Simple6502.html) has an external clock generator but feeds Φ₂out to the rest of the system. And I also had a look a the [VIC-20 schematic](http://www.zimmers.net/anonftp/pub/cbm/schematics/computers/vic20/index.html) and it too seems to use Φ₂out.

tebl, 2021-08-17:
Garth Wilson's 6502 primer is often the ones referred to on the 6502.org
forums, relevant sections:
http://wilsonminesco.com/6502primer/addr_decoding.html
http://wilsonminesco.com/6502primer/ClkGen.html

Apple1 Replica creation book:
https://www.applefritter.com/replica/chapter7

I agree that it is an outdated way of doing it, but it is an outdated
design which some may consider plugging very early chips into. The only
valid argument for keeping it this way, is that I consider the book to be
the documentation for it.

The board should perhaps have a wire selection for using the same clock as
the wdc 65c02 is mentioned on the silkscreen, but I never built a working
board with it. Would probably need a cleanup either way.