tebl/RC6502-Apple-1-Replica #29: Wrong jumper info in documentation

ea4eoz, 2021-11-02:
Hello!

I'm currently building the SBC version and I have noticed a problem in the description of two of the jumpers

The documentation (readme.md) says:

> _VP GND:
> 
> Shorted: WD65C02 CPU. Leaves pin 1 of the CPU (V̅P̅ on the WD65C02) floating.
> Open: Original MOS 6502, UM6502, SY6502, R65C02. Connects pin 1 of the CPU to 
> ground, providing an additional ground connection for the CPU._

In the revision H board it is just the opposite: Jumper open left pin 1 floating and jumper closed ties it to ground.

Another problem is with RAM Enable jumper:

> _RAM_EN
> 
> Left pair shorted: Enable RAM. Connects RAM C̅S̅/O̅E̅ to address decoding logic (CS_RAM).
> Right pair shorted: Disable RAM. Connects RAM C̅S̅/O̅E̅ to Vcc._

In revision H board the jumper is placed vertically, so it must be placed in the upper pair to enable the RAM and in the lower pair to disable it.

Hope it helps. Greetings!

0cjs, 2021-12-07:
@ea4eoz Thanks for your report! I'm sorry I took so long to have a look at it. Feel free to have a look at PR #33 and see if that fixes your issues.

ea4eoz, 2021-12-11:
Yes, now it looks fine. Thank you!

0cjs, 2021-12-18:
@ea4eoz I'm glad to hear that the problem is fixed. You should close the issue to indicate that. (Because you created this issue, you'll see the "Close issue" button beside the "Comment" button underneath the comment box.)

ea4eoz, 2021-12-18:
Ups... I missed the close button.

Thank you!

closed 2021-12-18 by ea4eoz