tebl/RC6502-Apple-1-Replica #31: Stuck terminal and die

sergiomtzlosa, 2021-11-11:
Hi!

I am trying since a coupe of months to get the Apple1 SBC (Rev F) work but no luck, the terminal only shows "RC6502 apple 1 Replica" and got stuck, no prompt at all 😓

[image]

I made some checks on the capacitor and they are all fine, also I reflash the Arduino and the 8KB ROM onto the AT28C64 but still no luck, please can you point me some other checks to find the error out please?

[image]
[image]

Thanks and stay safe!

jhow0672, 2021-11-18:
What tripped me up most was the jumper pins - double-check those. The instructions weren’t too clear. I built a version “H” board which is a little different.  Also, used MCP23S17 not MCP23O17.

sergiomtzlosa, 2021-11-18:
@jhow0672 Thank yo so much !! I will dig on that a little deeper and I will let you know my results.

Regards

0cjs, 2021-12-07:
> Also, used MCP23S17 not MCP23O17.

Good catch on that: that's almost certainly the problem. According to the [datasheet][ds], the `S` version is an [SPI] interface, the `O` version is [I²C]. These are not compatible!

That chip not working as expected is consistent with the symptoms: you can see messages from the Nano itself (the `RC6502 Apple 1 Replica` banner) but, because the Nano can't talk to the I/O expander, it can't talk through it to the PIA and so cannot communicate with the 6502 CPU.

[ds]: https://ww1.microchip.com/downloads/en/devicedoc/20001952c.pdf
[SPI]: https://en.wikipedia.org/wiki/Serial_Peripheral_Interface
[I²C]: https://en.wikipedia.org/wiki/I%C2%B2C

sergiomtzlosa, 2021-12-07:
Hi!!

Thank you so much for this amazing inputs!!!! I change the MCP, but I still do not get thw WOZMon, I checked the capacitor, they seem fine, please can you give some advice about out to place the jumpers?

[image]

I also connect thru screen with this command: screen /dev/ttyUSB0 115200

Is the SPI speed correct?

Thanks

0cjs, 2021-12-07:
If you see the banner from the Nano when you reset it, you have the speed correct. Note that's not the SPI interface speed, that's the serial interface speed (on the "other side" of the Nano).

So you replaced the MCP23O17 with an MCP23S17? (I can't see in the photograph at that resolution.) And you did press the board's reset button after seeing the Nano's banner, right?

If youv'e done all that, and still see no `\` prompt or response to input (particularly Esc), the next step is to start probing the board, I suppose. Do you have an oscilloscope? If not, do you have a logic probe? I guess if worst comes to worst you can use a volt meter. My idea is to start by probing the keyboard input and screen output pins of the MC6821 to see if you see the right things there when pressing a key and from what the last character output by the computer should be.

sergiomtzlosa, 2021-12-07:
Hi @0cjs 

I will do more deep reseach, I do not have an oscilloscope, just a volt meter....when I get some result I will be back

Thanks