tebl/RC6502-Apple-1-Replica #35: RC6502 Apple 1 SBC using MC68B21 causes unresponsiveness

denjhang, 2022-01-13:
[image]

denjhang, 2022-01-13:
Pressing the reset key on the sbc will bring up a "/", nothing happens to keyboard input, it seems to freeze.

denjhang, 2022-01-13:
Press the reset button on the arduino nano, "RC6502 Apple 1 Replica" will appear

0cjs, 2022-01-18:
The `RC6502 Apple 1 Replica` message is from the Nano, and that just indicates that the Nano is talking to the PC and thinks its talking to the SBC. Resetting that is entirely independent of resetting the SBC.

The `\` prompt is from the SBC; that's how WozMon says that you can start typing input. So if you're seeing that, you know that the Nano is reading the "screen" output from the SBC. If typing doesn't echo the characters you've typed (or a new prompt if you hit `Return` or `Esc`), it could be that your typing is not getting to the Nano, but more likely there's a problem with the connection between the Nano and the SBC's keyboard circuitry.

denjhang, 2022-01-19:
I've replaced a few sets of chips and the situation is still the same, it doesn't seem to be a chip problem. Can anyone tell me what to do?

0cjs, 2022-01-20:
I guess the next step might be to put a 'scope or logic probe on the "keyboard" outputs from the Nano to the PIA and make sure it's toggling those lines when you type at the console?

denjhang, 2022-01-30:
At the moment I found out that the cause is a bug caused by corrupted SRAM.

explit7, 2023-11-23:
I have exact the same Problem. Will try to replace the RAM