tebl/RC6502-Apple-1-Replica #36: Trouble with programming arduino

C-JESTER01, 2022-01-22:
Hello,
I recently bought a kit through https://retrotechlyfe.com and I had accidentally formatted the preprogrammed arduino that came with the kit. I would like some help figuring out which sketch I need to upload to the arduino, since the seller will not release any info. 

Board Rev: H
Controller: Arduino Nano 
Chip: ATmega328p

Thanks to who ever decides to help me out!

tebl, 2022-01-22:
You will find it at
https://raw.githubusercontent.com/tebl/RC6502-Apple-1-Replica/master/software/arduino/PIA%20Communicator/pia_communicator/pia_communicator.ino

C-JESTER01, 2022-01-22:
That is the program I found however, when I put it into the arduino IDE, it
fails to compile. Any tips?

tebl, 2022-01-22:
I think you'd need to be a bit more specific, which error are you getting?
Preferably the stack trace itself.

Usually it's just a missing library, but it's been a while since I last
looked at it - so there might've been changes somewhere.

C-JESTER01, 2022-01-22:
Yes it was missing MCP23S17.h, even though it was included. I changed the
path in the program and then it said it was unable to be compiled for that
board. Even though it is an Arduino Nano

C-JESTER01, 2022-01-22:
Give me a half hour to pull the error for you

C-JESTER01, 2022-01-22:
Arduino: 1.8.19 (Windows 10), Board: "Arduino Nano, ATmega328P"

C:\Users\Hunter\AppData\Local\Temp\ccCysob2.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_pia_communicator.ino.cpp.o.1987':

<artificial>:(.text.startup+0x64): undefined reference to `MCP23S17::MCP23S17(SPIClass*, unsigned char, unsigned char)'

C:\Users\Hunter\AppData\Local\Temp\ccCysob2.ltrans0.ltrans.o: In function `configure_bridge':

C:\Users\Hunter\Desktop\PIAV2\pia_communicator/pia_communicator.ino:41: undefined reference to `MCP23S17::begin()'

C:\Users\Hunter\Desktop\PIAV2\pia_communicator/pia_communicator.ino:45: undefined reference to `MCP23S17::pinMode(unsigned char, unsigned char)'

C:\Users\Hunter\Desktop\PIAV2\pia_communicator/pia_communicator.ino:47: undefined reference to `MCP23S17::pinMode(unsigned char, unsigned char)'

C:\Users\Hunter\Desktop\PIAV2\pia_communicator/pia_communicator.ino:51: undefined reference to `MCP23S17::pinMode(unsigned char, unsigned char)'

C:\Users\Hunter\AppData\Local\Temp\ccCysob2.ltrans0.ltrans.o: In function `output_status':

C:\Users\Hunter\Desktop\PIAV2\pia_communicator/pia_communicator.ino:59: undefined reference to `MCP23S17::readPort(unsigned char)'

C:\Users\Hunter\AppData\Local\Temp\ccCysob2.ltrans0.ltrans.o: In function `pia_send':

C:\Users\Hunter\Desktop\PIAV2\pia_communicator/pia_communicator.ino:104: undefined reference to `MCP23S17::writePort(unsigned char, unsigned char)'

C:\Users\Hunter\AppData\Local\Temp\ccCysob2.ltrans0.ltrans.o: In function `serial_transmit':

C:\Users\Hunter\Desktop\PIAV2\pia_communicator/pia_communicator.ino:156: undefined reference to `MCP23S17::readPort(unsigned char)'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board Arduino Nano.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tebl, 2022-01-22:
Have not tested it, but it seems that the library class has changed name -
try my older copy of the library at
https://github.com/tebl/RC6502-Apple-1-Replica/tree/master/software/arduino/PIA%20Communicator/lib/MCP23S17
.

Alternatively, you can try changing the text MCP23S17 in the sketch to MCP
(excluding the include statement at the top). It might work, but then again
I do not know how much has been altered other than then name.

C-JESTER01, 2022-01-23:
Arduino: 1.8.19 (Windows 10), Board: "Arduino Nano, ATmega328P"

pia_communicator:24:31: error: no matching function for call to
'MCP23S17::MCP23S17(SPIClass*, int, int)'

 MCP23S17 bridge(&SPI, IO_SS, 0);

                               ^

In file included from C:\Users\Hunter\Desktop\RC6502-Apple-1-Replica-master
- Copy\software\arduino\PIA
Communicator\pia_communicator\pia_communicator.ino:2:0:

C:\Users\Hunter\Documents\Arduino\libraries\MCP23S17/MCP23S17.h:31:3: note:
candidate: MCP23S17::MCP23S17(uint8_t, uint8_t)

   MCP23S17(uint8_t select, uint8_t address = 0x00);

   ^~~~~~~~

C:\Users\Hunter\Documents\Arduino\libraries\MCP23S17/MCP23S17.h:31:3:
note:   candidate expects 2 arguments, 3 provided

C:\Users\Hunter\Documents\Arduino\libraries\MCP23S17/MCP23S17.h:30:3: note:
candidate: MCP23S17::MCP23S17(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)

   MCP23S17(uint8_t select, uint8_t dataIn, uint8_t dataOut, uint8_t clock,
uint8_t address = 0x00);

   ^~~~~~~~

C:\Users\Hunter\Documents\Arduino\libraries\MCP23S17/MCP23S17.h:30:3:
note:   candidate expects 5 arguments, 3 provided

C:\Users\Hunter\Documents\Arduino\libraries\MCP23S17/MCP23S17.h:27:7: note:
candidate: constexpr MCP23S17::MCP23S17(const MCP23S17&)

 class MCP23S17

       ^~~~~~~~

C:\Users\Hunter\Documents\Arduino\libraries\MCP23S17/MCP23S17.h:27:7:
note:   candidate expects 1 argument, 3 provided

C:\Users\Hunter\Documents\Arduino\libraries\MCP23S17/MCP23S17.h:27:7: note:
candidate: constexpr MCP23S17::MCP23S17(MCP23S17&&)

C:\Users\Hunter\Documents\Arduino\libraries\MCP23S17/MCP23S17.h:27:7:
note:   candidate expects 1 argument, 3 provided

C:\Users\Hunter\Desktop\RC6502-Apple-1-Replica-master -
Copy\software\arduino\PIA
Communicator\pia_communicator\pia_communicator.ino: In function 'void
output_status()':

pia_communicator:59:37: error: 'class MCP23S17' has no member named
'readPort'; did you mean 'read8'?

   debug_value("Video D0-D6", bridge.readPort(IO_VIDEO) & 127);

                                     ^~~~~~~~

                                     read8

C:\Users\Hunter\Desktop\RC6502-Apple-1-Replica-master -
Copy\software\arduino\PIA
Communicator\pia_communicator\pia_communicator.ino: In function 'void
pia_send(int)':

pia_communicator:104:12: error: 'class MCP23S17' has no member named
'writePort'; did you mean 'write8'?

     bridge.writePort(IO_KBD, c | 128);

            ^~~~~~~~~

            write8

C:\Users\Hunter\Desktop\RC6502-Apple-1-Replica-master -
Copy\software\arduino\PIA
Communicator\pia_communicator\pia_communicator.ino: In function 'void
serial_transmit()':

pia_communicator:156:21: error: 'class MCP23S17' has no member named
'readPort'; did you mean 'read8'?

     char c = bridge.readPort(IO_VIDEO) & 127;

                     ^~~~~~~~

                     read8

exit status 1

no matching function for call to 'MCP23S17::MCP23S17(SPIClass*, int, int)'

I am now getting this error. I see that it suggests to change some commands
to a different one (Like readPort to read8). Should I try that or did I
mess something else up?

Thank you for your help thus far

countach22, 2022-02-28:
I am getting the same error while trying to verify/compile the PIA Communicator Sketch. I have tried this on two different windows machines and a mac.  Has anyone figured out the source of the error yet?  Thanks.

Arduino: 1.8.19 (Mac OS X), Board: "Arduino Nano, ATmega328P"

PIA_Communicator_old:26:31: error: no matching function for call to 'MCP23S17::MCP23S17(SPIClass*, int, int)'
 MCP23S17 bridge(&SPI, IO_SS, 0);
                               ^
In file included from /Users/Erik/Documents/Arduino/PIA_Communicator_old/PIA_Communicator_old.ino:4:0:
/Users/Erik/Documents/Arduino/libraries/MCP23S17/MCP23S17.h:31:3: note: candidate: MCP23S17::MCP23S17(uint8_t, uint8_t)
   MCP23S17(uint8_t select, uint8_t address = 0x00);
   ^~~~~~~~
/Users/Erik/Documents/Arduino/libraries/MCP23S17/MCP23S17.h:31:3: note:   candidate expects 2 arguments, 3 provided
/Users/Erik/Documents/Arduino/libraries/MCP23S17/MCP23S17.h:30:3: note: candidate: MCP23S17::MCP23S17(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)
   MCP23S17(uint8_t select, uint8_t dataIn, uint8_t dataOut, uint8_t clock, uint8_t address = 0x00);
   ^~~~~~~~
/Users/Erik/Documents/Arduino/libraries/MCP23S17/MCP23S17.h:30:3: note:   candidate expects 5 arguments, 3 provided
/Users/Erik/Documents/Arduino/libraries/MCP23S17/MCP23S17.h:27:7: note: candidate: constexpr MCP23S17::MCP23S17(const MCP23S17&)
 class MCP23S17
       ^~~~~~~~
/Users/Erik/Documents/Arduino/libraries/MCP23S17/MCP23S17.h:27:7: note:   candidate expects 1 argument, 3 provided
/Users/Erik/Documents/Arduino/libraries/MCP23S17/MCP23S17.h:27:7: note: candidate: constexpr MCP23S17::MCP23S17(MCP23S17&&)
/Users/Erik/Documents/Arduino/libraries/MCP23S17/MCP23S17.h:27:7: note:   candidate expects 1 argument, 3 provided
/Users/Erik/Documents/Arduino/PIA_Communicator_old/PIA_Communicator_old.ino: In function 'void output_status()':
PIA_Communicator_old:61:37: error: 'class MCP23S17' has no member named 'readPort'; did you mean 'read8'?
   debug_value("Video D0-D6", bridge.readPort(IO_VIDEO) & 127);
                                     ^~~~~~~~
                                     read8
/Users/Erik/Documents/Arduino/PIA_Communicator_old/PIA_Communicator_old.ino: In function 'void pia_send(int)':
PIA_Communicator_old:106:12: error: 'class MCP23S17' has no member named 'writePort'; did you mean 'write8'?
     bridge.writePort(IO_KBD, c | 128);
            ^~~~~~~~~
            write8
/Users/Erik/Documents/Arduino/PIA_Communicator_old/PIA_Communicator_old.ino: In function 'void serial_transmit()':
PIA_Communicator_old:158:21: error: 'class MCP23S17' has no member named 'readPort'; did you mean 'read8'?
     char c = bridge.readPort(IO_VIDEO) & 127;
                     ^~~~~~~~
                     read8
exit status 1
no matching function for call to 'MCP23S17::MCP23S17(SPIClass*, int, int)'

countach22, 2022-03-05:
Despite knowing very little about linux, I managed to compile the pia_communicator C code (by J.B. Langston) to a hex file (pia.hex) and successfully uploaded it to the nano.  Now the Apple-1 works and  I am using machine language in the WozMon !

I have attached the hex file.  It needs to be uploaded to the nano from a windows command line interface using avrdude. 

Instructions here : https://arduino.stackexchange.com/questions/60599/how-can-i-upload-a-hex-file-to-an-arduino-uno

[pia.hex.zip](https://github.com/tebl/RC6502-Apple-1-Replica/files/8189893/pia.hex.zip)

> Hello, I recently bought a kit through https://retrotechlyfe.com and I had accidentally formatted the preprogrammed arduino that came with the kit. I would like some help figuring out which sketch I [...]

giobbino, 2022-09-09:
> Have not tested it, but it seems that the library class has changed name - try my older copy of the library at https://github.com/tebl/RC6502-Apple-1-Replica/tree/master/software/arduino/PIA%20Communicator/lib/MCP23S17 .

I copied the two files mcp23s17.cpp and mcp23s17.h  from the above link to the folder pia_communicator (where there is the pia_communicator.ino file).

Then I changed in both pia_communicator.ino and mcp23s.cpp the sentence 
from:  **#include <mcp23s17.h>** 
    to:  **#include "mcp23s17.h"**

and finally Arduino IDE compiled it (without to change it, it doesn't compile, at least for me. I suppose it takes the mcp23s17.h from another place).

Since I haven't right now an Arduino Nano, I uploaded to the LG8F328P mini EVB, that should be a drop-in Nano replacement (see https://forum.hobbycomponents.com/viewtopic.php?t=2994). My board is the purple model. 

I works, more or less: I got the "APPLE I REPLICA" welcome text on PuttY @115.2 kbps, but then every key pressed returns just a "?" or "_" echo on the screen.

Since I don't know anything about Apple I, Woz Mon etc., I'm not sure what should I expect from it, but it seems to me there's something wrong...

countach22, 2022-09-12:
Maybe this video would help?
https://youtu.be/jSqrTGpBXsE

> Since I don't know anything about Apple I, Woz Mon etc., I'm not sure what should I expect from it, but it seems to me there's something wrong... maybe the board I'm using isn't fully compatible? It should be, but who knows...

giobbino, 2022-09-12:
> Maybe this video would help?
> https://youtu.be/jSqrTGpBXsE

It will, thank you... but before all I need to understand why just "_"s and "?"s appear on the screen, whatever I type on the keyboard. I got the welcome text, so I suppose it can talk properly with the terminal program (Putty). I must to test with a legit Arduino Nano (instead of the drop-in replacement board I'm using), but I haven't any right now.
I'm also making an U-Term board but I'm waiting for some parts yet...
Will tell you in a few weeks...