tebl/RC6502-Apple-1-Replica #37: Problems getting color VDU working

StormTrooper, 2022-02-13:
I've built a color VDU and double checked all the connections and soldering joints. 
When I plug it all in I get either a red screen of G's or red G's with some random green characters. 

This seems to suggest at least the video chip is working. I've tested all logic chips and alsl tested the 62256 ram chip - no issues there. 

[image]

[image]

Any ideas as to where to start troubleshooting on this?

The RC6502 is running fine and I can connect to the console via the Arduino and run basic programs.

0cjs, 2022-06-05:
In case it's not clear (because I think another [another user](https://github.com/tebl/RC6502-Apple-1-Replica/issues/42) may also have been unclear on this), the VDU board simply gives you a frame buffer in memory: you are responsible for writing the software that intialises it, prints text on the screen, and so on. Especially, the Apple 1 monitor (WOZMON) normally used with this system will _not_ print anything on this display; the Apple 1 video system is very different from this, and does not even involve a frame buffer. (Apple 1 video is more like a terminal.)

StormTrooper, 2022-06-05:
Thanks, very useful info linked above. I haven't had any time to further investigate this but that will be the cause then - no software has been done to initialize the VDU board. 

Will close this issue as it's a lot clearer now.

closed 2022-06-05 by StormTrooper