tebl/RC6502-Apple-1-Replica #38: Wrong IC label on Serial IO Board

nebanks, 2022-03-16:
U3 on the Serial IO Board is labeled on the schematic (and the silkscreen) as 74HCT38. It should definitely be 74HCT**1**38.

0cjs, 2022-06-05:
I've just checked my two-year-old PDF copy of the schematic and [the latest exported schematic][sch]
and in both cases it's marked `74HCT138`. It's also correct in this (five year old) [exported image of the board][img]. Can you tell us exactly which file you're looking at, which commit it came from, and confirm that the problem still exists in the latest commit?

[sch]: https://github.com/tebl/RC6502-Apple-1-Replica/blob/master/RC6502%20Apple%201%20SBC/export/RC6502%20Apple%201%20SBC.pdf
[img]: https://github.com/tebl/RC6502-Apple-1-Replica/blob/master/RC6502%20Apple%201%20SBC/gallery/2017-07-09%2015.17.02.jpg

nebanks, 2022-06-05:
Again, you are looking at the SBC, and not the Serial I/O Board schematic. The schematics I'm looking at are always the most current ones in the Git repo. This particular one is listed as "Rev B:" 

[https://github.com/tebl/RC6502-Apple-1-Replica/blob/master/RC6502%20Serial%20IO/export/RC6502%20Serial%20IO.pdf](url)