tebl/RC6502-Apple-1-Replica #40: WD65C02 Does not work with CPU Module

nebanks, 2022-04-04:
The schematic and the silkscreen for the CPU Module PCB list U1 as "WD65C02." That is just the one processor that WILL NOT WORK with the CPU module; the BE pin on the CPU module is left unconnected, but must be pulled high in order for the WD65C02 to work.

0cjs, 2022-06-05:
On my oldish copy of the schematic (no newer than 2020-01-10) pin 36 of U1 is marked `BE` and is connected to VCC through a 3.3k resistor, R10. I've also just checked the [latest exported schematic][sch] and, though it's very different in layout, it also has the same pull-up on that pin. Perhaps you have a (very) old copy of the schematic? At any rate, telling us exactly what file from which commit you are looking at would be very helpful.

(Unfortunately the schematic revision indication doesn't seem to be getting bumped with new revisions; both mine and the latest one above say "Rev: B" though they are clearly different. @tebl, maybe you can try to remember to update that when you update the schematic?)

[sch]: https://github.com/tebl/RC6502-Apple-1-Replica/blob/master/RC6502%20Apple%201%20SBC/export/RC6502%20Apple%201%20SBC.pdf

nebanks, 2022-06-05:
Hi Curt,

I think you are talking about the SBC, which does have the required pull-up resistor. I'm talking about the CPU module. The current schematic is marked "rev B."

0cjs, 2022-06-05:
Right, I think now perhaps you're seeing why it would be very helpful for you to _specify the file you're looking at,_ rather than just saying things like "CPU module." You might get a much faster response on these bugs if you'd do that.

nebanks, 2022-06-05:
The name if the file is CPU module.

0cjs, 2022-06-05:
Well I am interested in fixing things like this in the repo; you'll see I have close to a dozen commits in there already. However, I'm not interested in working on issues from people who insist that they're being clear when refusing even to give a proper filename (there is no file in the entire repo named `CPU`) or link to the GitHub page for that file.

I'm guessing you're talking about some file under the `RC6502 CPU/` directory, but I can't be bothered to try to chase down which one you were looking at. (Keep in mind that the Gerber exports, PDF exports and image files may not contain the same information as the schematic files or each other, for example.)

I'm unsubscribing from this issue; I hope you find someone to fix it for you.

nebanks, 2022-06-05:
You do realize that the RC6502 is a *modular backplane computer* that happens to have a self-contained SBC module? Each module has a separate directory named "CPU" for the CPU module, "Serial I/O" for the Serial I/O module, etc. The schematic for the CPU module - which is named "CPU" - can be found, as you astutely surmise, in the "CPU" subdirectory. (In fact, in the "export" sub-subdirectory, the same as every other schematic for every other part of the project.) This issue has been "fixed" by me, in the sense that I soldered a pull-up resistor between BE and  VCC on the CPU board. No one else I've talked to about this has had any trouble getting the point, including TEBL himself. I guess it's up to him if he wants to change the design so that BE is not left floating. 

Sorry if you're having a bad day, or whatever, but I'm not giving you my lunch money.

tebl, 2022-06-05:
Yes, you are right - the modular boards have not seen much work as I've
gotten distracted by other things (currently at the hospital with a 3 day
old baby, but on the best of days I am not healthy enough to work
consistently). These projects are on a hobby basis and require voluntary
contributions such as the one Curt is providing by ensuring that most can
get a response.

I have a different version of the CPU module on my desk somewhere that adds
the missing resistor, but I need to test it before putting it out as there
was a bit of rerouting done. So it is somewhere in the future plans, I just
need to work up to being allowed to have separate plans 😅