tebl/RC6502-Apple-1-Replica #42: IMO it lacks some detailed info (or a tutorial)

giobbino, 2022-05-29:
Hi all,
I found this awesome project and I decided to give it a try; I bought the needed PCBs on PCBway and I'm waiting for them.
However, sice I don't want to depend on my PC (using Putty etc.) I was trying to understand how to connect a (PS/2?) keyboard and a composite monitor. I got the SBC, backplane, VDU and vdu adapter boards.

I can't find any info about that. Is there any tutorial, page, youtube video about that?

cheers from Brazil :-)

0cjs, 2022-06-05:
If what you're looking for is simple setup for a terminal you can type at, using a VDC board on this system is probably not the way to go. It seems there's a fair bit of work to be done here, possibly more than you'd care to take on at your current level of expertise. There are several issues here.

First, I am guessing (I've not used it) that the 6847 VDU should display text just fine so long as you put the right character codes into the frame buffer memory. However, you'd need software to do this, including keeping track of the cursor position, scrolling the screen, handling control characters, and so on. The Apple 1 monitor (WOZMON) typically used with this board does not have any code to support anything like this; you'd need to write your own character output routines. (The Apple 1 video design is _very_ different from this design.) You can look at the BIOSes of many different 1980s microcomputers (the Apple II BIOS is one example) to get a sense of how this code should work. It's not terribly difficult, but it's not a beginner project.

Second, there's no keyboard interface here for PS/2 keyboards as far as I'm aware, or really any interface that's "officially" for a keyboard. One relatively easy option (if you have the technical skill and familiarity with how these things work) is to grab whatever matrix-based keyboard you might have handy, hook it up to a [6820 or similar PIA][pia] (you could even re-purpose the one already on the SBC for this if you don't mind losing the interface to the Arduino Nano) and use that to scan the keyboard in traditional '80s style. Again, lots of old BIOSes have code for this, and I even happen to have recently [reverse-engineered the C16 keyboard hardware and software][c16], which should give a pretty decent sense of how these things work. (Again, if you have sufficient technical skills to understand what's going on there.)

If you're not up to doing this level of work, I'd strongly suggest you stick with the SBC's basic configuration of having the Arduino Nano emulate the screen and keyboard hardware of the original Apple 1. You can then use that system to learn what you need to know to develop a video display and keyboard system. You might start by going through the Apple 1 WOZMON listing and Apple 1 or SBC schematic and learning how all that works; if that doesn't seem reasonably simple to you yet, you're probably not (yet!) at the technical level to be writing code to handle running a text video display.

[pia]: https://en.wikipedia.org/wiki/Peripheral_Interface_Adapter
[c16]: https://gitlab.com/retroabandon/cbm/-/blob/master/c16/keyboard.md

0cjs, 2022-06-05:
Also, it would be nice if someone could change the title of this issue to something more descriptive, such as "How can I use a text video display and keyboard on the system?" The current title is going to be rather mysterious when it comes up in response to someone searching the issues.

giobbino, 2022-06-05:
> Also, it would be nice if someone could change the title of this issue to something more descriptive, such as "How can I use a text video display and keyboard on the system?" The current title is going to be rather mysterious when it comes up in response to someone searching the issues.

that was the easiest part :D

giobbino, 2022-06-05:
> If what you're looking for is simple setup for a terminal you can type at, using a VDC board on this system is probably not the way to go. It seems there's a fair bit of work to be done here, possibly more than you'd care to take on at your current level of expertise. There are several issues here.

Thank you for your (rather good!) explanation. I was maybe too naive supposing it was an Apple I clone/replica with the CRT terminal output too. I also supposed it  could be possible to hook up a ps/2 keyboard through the Arduino Nano. My fault, should have studied the project better. Probably I was too impulsive, since I haven't enough skill to code at such low level. 

Oh, well, maybe somebody in the future will develope some CRT and keyboard routines...

0cjs, 2022-06-05:
I should mention that using the Nano serial interface is very close to using an original Apple 1 anyway. As I mentioned, the Apple 1's display is basically a (_very_) dumb terminal: you write a character to the 6820 output port B and the display circuitry picks that up and prints it. The only control character it knows is ASCII CR (carriage return) which moves the cursor to the start of the next line; you cannot even backspace or clear the screen. Input from the keyboard is similar: you read from port A of the PIA and get an ASCII character generated by the keyboard. So using a serial terminal is basically no different at all, as far as the software is concerned, and very little different if you run your terminal emulator in a 40×24 window on your PC. (I suppose you could program the Nano or some other microcontroller to generate a video display and read a physical keyboard, doing the same emulation of the Apple 1's video/keyboard hardware to the 6850, but I don't see a whole lot of point to that.)

The "BIOS" (WOZMON) provide any support for I/O from/to any other devices (there's no room for that in 256 bytes!), so most Apple 1 software is going to use those screen output and keyboard input routines or just directly read and write the hardware in the same way that WOZMON does.

So if you're just looking to run Apple 1 programs, you already have a system that's basically the same as the real thing; just set your terminal window to 40×24. Anything beyond that is going to involve programming (the Apple 1 pretty much was a, "here's the hardware, write whatever software you need" system), which is also essentially what using a contemporary machine was like.

giobbino, 2022-06-05:
Got it. What I wrongly expected was an emulation of a CRT terminal by the VDU, to avoid the use of a PC software... more or less something like the Hansel's Altair 8800 clone and the VDM-1 emulator ([https://github.com/dhansel/VDM1/blob/master/README.md]). 

In my mind the Apple 1 was no much more than an Altair with an embedded CRT terminal output, a tape recorder port, a keyboard interface and some ROM, and of course without the Altair front panel.
Must to be said the Hansel's project comes with a big bunch of software on an SD card that in the '70s you must to load teletype "typing" the bootloader by the panel switches; however when you've loaded the software (i.e. the Altair Basic), it runs on the CRT terminal (I mean: the composite or VGA output).

However I wish to stress that I'm not complaining about this project; obviously it was my mistake about what I expected vs. what the project offers.

I will build it and try to dig a little more and see if I'll be able to learn something more about it  :-)

0cjs, 2022-06-07:
@giobbino I think the issue may just be a technical misunderstanding about the Apple 1 interface. Don't think of it as a computer with a VDU and keyboard; think of it as a _really_ dumb terminal and a computer that happen to be on the same PCB. It's not nearly as flexible or powerful as an Altair with a serial interface and a terminal attached because though the I/O works the same from the computer's point of view (send characters out one port, receive them on another), with an Altair you can use a smarter terminal and get sophisticated functions like "backspace."

The real point of the Apple 1 was to give you something like an Altair (or more like an SWTPC, really) with a really cheap terminal. And the RC6502 SBC very much does that, slightly adapted to the modern reality that a terminal program running on your existing PC is cheaper than buying a keyboard and a video display these days.

giobbino, 2022-06-07:
Roger on that. I believed the VDU board was exactly a modern replica of the video section on the original Apple I (the video section on the Apple I sort of works even without the logic board populated, of course just displaying some garbage until you send a reset command, IIRC; since I've seen a similar video garbage in some screenshots here on this forum, I mistakenly assumed the VDU card was the same of the original video section).

Looking the whole thing from another point of view, is there any way to make the VDU board working like the original video section of the Apple I ? What's missing to complete this step? AFAIK the original video section had just a small character ROM and it was driven by the logic glue of the CPU section. Any chance to build the same "link" between the two sections?

Of course, I didn't mention the keyboard yet LOL

0cjs, 2022-06-08:
The Apple 1 is not "driven by the logic glue of the CPU section"; the two are completely separate systems (albeit on a single PCB). The "link" between the two sections is already on the SBC: it's the 6820 PIA. That provides essentially a standard 7-bit terminal interface except that it's parallel instead of serial. The Arduino is responsible for simulating _all_ of the Apple 1's video circuitry and an ASCII keyboard, turning this parallel interface into a serial terminal connection.

If you want to hook up an actual serial terminal that's quite trivial. The Nano, like the original Arduino Uno, uses an ATmega MCU that provides serial transmit and receive on pins 1 (TX) and 0 (RX). The USB serial interface is a separate FTDI FT232RL chip that connects to those serial pins and converts them to USB serial. So you can simply ignore the USB on that board (except to power it—see below) and connect Nano pins 1 and 2 to the TTL serial interface on the board you suggested.

The Nano still does need +5V power. You can get that by plugging in a USB power supply (which can also supply the SBC via closing jumper J8) or, if the SBC is powered from the backplane, leaving the USB port unplugged and J8 closed will also power the Nano from the backplane. (You _must_ have J8 open if you plug USB of any kind into the Nano while the SBC is powered from the backplane.) You could probably even plug the USB into a computer and see the output on both a computer terminal emulator and the actual terminal at the same time. Just make sure you don't type on both at the same time or the serial input data to the Nano is likely to be corrupt.

This still won't quite reproduce the behaviour of the original Apple 1 video system with the current Arduino firmware. Most of that is easy to fix in the firmware itself since you basically just need to remove capabilities from your terminal. The new firmware should:
- Ignore all characters in the control sticks¹ ($00-$1F) except for CR ($0D), for which it should send CR-LF ($0D $0A) out the serial port
- Translate both lower-case sticks (including punctuation) to the corresponding character in the upper-case sticks, to conform to what's available in the [Apple 1 character set][a1cs].
- Insert a CR-LF after the 40th character is printed on a line, to emulate the wrap-around behaviour of the Apple 1 video system.

The only things that this doesn't emulate are the actual character glyphs from the character ROM and the 24-line display. The latter can be dealt with by using a 24-line terminal or making the window of your terminal emulator program shorter. Note that emulation of the slow output rate (60 CPS) is already done by the Nano itself, which won't indicate to the 6820 that it's ready to receive a character until 1/60 second has elapsed. (Or, at least, something is slowing it down to that rate for me.) So you can use pretty much any terminal speed you want on the serial line, so long as it's at least 600 bps.

That said, I'm not really sure that there's a whole lot of point to this. The Apple 1 video system was surely not at the time considered to be characteristic of the Apple 1, the way video is on other home computers; and in particular it wasn't a "full-screen" (addressable) video display like contemporary S-100 bus video boards or (later) PET, TRS-80 and Apple II offered. It was just a cheap way of getting characters in and out for those who couldn't afford a proper terminal, filling the same role as the [TV typewriter][tvt] might fill for someone who had an Altair or whatever. So using a terminal emulator on a PC is not at all inaccurate, and neither a PC terminal emulator nor a physical terminal are emulating the actual look of the Apple 1 video output and its characters.

__________
¹ A "stick" in ASCII is a 16-character group differing only in the low 4 bits, e.g. $00-$0F or `@` through `O`.

[a1cs]: https://en.wikipedia.org/wiki/Apple_I#Apple_I_character_set
[tvt]: https://en.wikipedia.org/wiki/TV_Typewriter

giobbino, 2022-06-08:
In first, I wish to say I really appreciate your big effort to clarify me the doubts I had (and still partially have). It took a good time to write such a clear and detailed explanation. Many, many thanks for that!!!

It seems to me the TX and RX pins from the Nano are present on the backplane (pin #36 and #37) so, unless if you use another module that rely on these signals (I didn't check all the modules so I don't know if there are any), a VT-100 clone could be made to fit the backplane connector. This way you can have an on-board terminal (keyboard and VGA/composite video out) and, if needed, the VDU module (how to use the VDU isn't clear to me yet, though).

I understand I can't have a 100% emulation of the original Apple I; but I love to build standalone things that don't rely on other pieces of hardware. Just to explain why I'm trying to avoid to hook up it to a PC - that would be the easier and more reasonable approach, of course :-)

giobbino, 2022-06-12:
I'm wondering if an easier (and cheaper) way could be an ESP32, like the TTGO module - [example](https://www.aliexpress.com/item/33014937190.html) with the [FABGL libraries](https://www.github.com/fdivitto/fabgl) by Fabrizio Di Vittorio, that also come with terminal class and samples. 

One could use the small ESP32 module only and build a PCB to fit the backplane, adding PS2 and VGA (or composite?) output, or simply buy the whole module shown in the above link, with all the needed connectors, and link it by wire to the Arduino TX/RX pins. 

The ESP32 has an embedded wifi module that could (maybe) used for uploading software from PC/internet through the serial port.

Am I missing something?

0cjs, 2022-06-13:
Easier and cheaper than what? The [uTerm](https://hackaday.io/project/165325-uterm) seems to do pretty much what you want, [the PCB](https://www.pcbway.com/project/shareproject/uTerm__a_VT100_like_terminal_for_the_Z80_MBC2_.html) is $5, the STM chip it uses is about $2, and IIRC the rest of the BOM is a few dollars or so (though in my case it's all parts that I already had in my parts box).

Anyway, though the SBC is called a "replica," I think it's more accurate to call it a "clone": it gives effectively the same functionality as an Apple 1, but while the computer side of it is essentially the same design with different details (static RAM and more modern decoding), the video/keyboard side is a completely different implementation that gives similar functionality on a serial terminal, rather than a video display.

giobbino, 2022-06-13:
> Easier and cheaper than what? The [uTerm](https://hackaday.io/project/165325-uterm) seems to do pretty much what you want, [the [...]

Great advice, I didn't know the uTerm project. In term of cost, I was comparing the ESP32 vs. the Geoff's VT-100-like ASCII terminal; it uses a PIC32 that's quite expensive (~25-30 USD, considering shipping to Brazil, plus PCB etc), while an ESP32 costs less than 20 USD including shipping. 

The uTerm seems quite cheaper too, maybe just a little more expensive than an ESP32; the shipping fee for PCB and STM32 chip is the worst part, but I have all the remaining components. 

I already own a spare ESP32 to play a bit, but I think I will follow your advice and build a uTerm board. Thank you!

tebl, 2022-10-11:
Yes, I think FabGL would definitely be a way to go and I've had a
browser-tab open for it for two years hoping to get around to it 😅

The RC6502 was mainly my starting point, building a lot of extra modules
that never had much in common with Apple 1 beyond what I wanted to build
for the fun of it. Besides the Arduino bits I've built things more from a
starting point of browsing old books and picking projects that way.

At the rate I'm going I'll probably have lost my remaining strands of hair
before I get around to more current technology, other than to suggest it
would be a fun project for anyone interested in such things - sounds like
an excellent opportunity for learning 😀